willprice/python-omxplayer-wrapper #115: Unable to locate package libdbus, ImportError: No module named dbus

caseyanderson, 2017-11-14:
# Issue Report
## ImportError: No module named dbus

## Problem reproduction
Ran the following: `sudo apt-get update && sudo apt-get install -y libdbus-1{,-dev}`

Which resulted in: `E: Unable to locate package libdbus-1`

Installed from pip as instructed in docs. both of the following lines result in this error:

* `import omxplayer.player`
* `from omxplayer.player import OMXPlayer`

## Environment details
* OS (`lsb_release -a`):
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `dpkg-query: package 'python-dbus' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.` |
| `python3` (`python3 --version`) | `Python 3.4.2` |
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

jehutting, 2017-11-14:
Use
```
sudo apt-get install libdbus-1-dev
```

willprice, 2017-11-14:
It looks like the `setup.py` is missing the dependency, which can now be added since `dbus` is now on pip... I'll get round to it sometime this week/weekend.

willprice, 2017-11-14:
@caseyanderson thanks for the report. I've just pushed a commit to `develop` that should fix this. Can you try again installing from pip:

`pip install -e git+https://github.com/willprice/python-omxplayer-wrapper.git@develop#egg=omxplayer`

flurohair, 2017-11-15:
@willprice 
Running this `pip install -e git+https://github.com/willprice/python-omxplayer-wrapper.git@develop#egg=omxplayer`

got 2 errors:
`  Failed building wheel for dbus-python`

`Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-eyo9UV/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ZkEc34-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-eyo9UV/dbus-python/
`

caseyanderson, 2017-11-15:
@willprice, just saw your note, pardon the delayed response. is it still helpful for me to try that line or do you have the information you need from @flurohair's comment above?

willprice, 2017-11-15:
Try running and and see if you get the error, if you do try installing the
dbus-python from your package manager

caseyanderson, 2017-11-15:
Running this on Raspbian Jessie Lite:
` pip3 install -e git+https://github.com/willprice/python-omxplayer-wrapper.git@develop#egg=omxplayer`

Results in this:
```Obtaining omxplayer from git+https://github.com/willprice/python-omxplayer-wrapper.git@develop#egg=omxplayer
  Updating ./src/omxplayer clone (to develop)
  Running setup.py (path:/home/pi/src/omxplayer/setup.py) egg_info for package omxplayer
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/home/pi/src/omxplayer/setup.py", line 3, in <module>
        import omxplayer
      File "/home/pi/src/omxplayer/omxplayer/__init__.py", line 1, in <module>
        from omxplayer.player import OMXPlayer
      File "/home/pi/src/omxplayer/omxplayer/player.py", line 18, in <module>
        from dbus import DBusException, Int64, String, ObjectPath
    ImportError: No module named 'dbus'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/home/pi/src/omxplayer/setup.py", line 3, in <module>

    import omxplayer

  File "/home/pi/src/omxplayer/omxplayer/__init__.py", line 1, in <module>

    from omxplayer.player import OMXPlayer

  File "/home/pi/src/omxplayer/omxplayer/player.py", line 18, in <module>

    from dbus import DBusException, Int64, String, ObjectPath

ImportError: No module named 'dbus'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /home/pi/src/omxplayer
Storing debug log for failure in /home/pi/.pip/pip.log```

Running this to install `dbus-python`: `pip3 install dbus-python`

Results in this:
```checking for DBUS... no

configure: error: Package requirements (dbus-1 >= 1.6) were not met:

No package 'dbus-1' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you

installed software in a non-standard prefix.

Alternatively, you may set the environment variables DBUS_CFLAGS

and DBUS_LIBS to avoid the need to call pkg-config.

See the pkg-config man page for more details.

Traceback (most recent call last):

  File "<string>", line 1, in <module>

  File "/tmp/pip-build-mr8d4vgd/dbus-python/setup.py", line 106, in <module>

    'build_ext': BuildExt,

  File "/usr/lib/python3.4/distutils/core.py", line 148, in setup

    dist.run_commands()

  File "/usr/lib/python3.4/distutils/dist.py", line 955, in run_commands

    self.run_command(cmd)

  File "/usr/lib/python3.4/distutils/dist.py", line 974, in run_command

    cmd_obj.run()

  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run

    return orig.install.run(self)

  File "/usr/lib/python3.4/distutils/command/install.py", line 583, in run

    self.run_command('build')

  File "/usr/lib/python3.4/distutils/cmd.py", line 313, in run_command

    self.distribution.run_command(command)

  File "/usr/lib/python3.4/distutils/dist.py", line 974, in run_command

    cmd_obj.run()

  File "/tmp/pip-build-mr8d4vgd/dbus-python/setup.py", line 62, in run

    cwd=builddir)

  File "/usr/lib/python3.4/subprocess.py", line 561, in check_call

    raise CalledProcessError(retcode, cmd)

subprocess.CalledProcessError: Command '['/tmp/pip-build-mr8d4vgd/dbus-python/configure', '--disable-maintainer-mode', 'PYTHON=/usr/bin/python3', '--prefix=/tmp/pip-build-mr8d4vgd/dbus-python/build/temp.linux-armv7l-3.4/prefix']' returned non-zero exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-mr8d4vgd/dbus-python/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-yfkem2ft-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-mr8d4vgd/dbus-python
Storing debug log for failure in /home/pi/.pip/pip.log```

Goyardbenjamin, 2017-11-26:
Hello, 

I tried to use your code without any  sucess on m'y raspberry pi 3 , with raspbian os.
I have the same issue explain in this topic.

My english is poor and i don't understand if you fix it ?

chelseamw13, 2017-12-08:
@caseyanderson were you ever able to get rid of the "ImportError: No module named 'dbus'? I have used this wrapper successfully before and am running into this problem when trying to install it on a new pi.

caseyanderson, 2017-12-08:
I gave up on the dbus route and instead have been controlling `omxplayer` from `python` via `subprocess`. So far so good. See [here](https://github.com/caseyanderson/pi_video_control) for very basic examples of what i have been doing along these lines.

willprice, 2017-12-08:
@chelseamw13 You need to install `python-dbus` from your package manager: `sudo apt install python-dbus` or pip: `sudo pip install dbus`

chelseamw13, 2017-12-08:
@willprice after running sudo apt install python-dbus i have version 1.2.0-2+b1 compatible with python 2.7 

 I believe i need version 1.2.4 and have tried your recommended installation process (sudo apt-get update && sudo apt-get install -y libdbus-1{,-dev}) but have had no luck. Just checked and unfortunately I'm still getting the same "no module named dbus" error!

a side note: with sudo pip install dbus i get the error "could not find a version that satisfies the requirement bus(from versions: ) No matching distribution found for dbus

chelseamw13, 2017-12-09:
"apt-get install libdbus-glib-1-dev"
 "pip3 install dbus-python"

seems like one of these two worked!

willprice, 2018-11-03:
The dependency is explicit as of commit ccf34d908de76c58402d9e9cfe6ad36ef2edb73a 
as @chelseamw13 says, you might need dependencies installed to build the dbus package:

```
$ sudo apt install libdbus-glib-1-dev
```

Failing that install the OS packaged version

```
$ sudo apt install python{,3}-dbus
```

closed 2018-11-03 by willprice