willprice/python-omxplayer-wrapper #117: No module named omxplayer.player

landtree, 2018-01-16:
# Issue Report
## Description
When running example code get error

Traceback (most recent call last):
  File "pyplayer.py", line 3, in <module>
    from omxplayer.player import OMXplayer
    ImportError: No module named omxplayer.player

## Problem reproduction

Running example local file code.

## Environment details
* OS (`lsb_release -a`):
```
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch

```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.5` |
| `python-dbus' |  1.2.4-1 |
| `python' | `2.7.13` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

-------

willprice, 2018-01-16:
How have you got the lib installed, how do you invoke your script, what is `sys.path` and `sys.executable`?

landtree, 2018-01-17:
I followed the issue thread for the libdbus error.

**Running:**
sudo pip install dbus

**I get:**
Collecting dbus
Could not find a version that satisfies the requirement dbus (from versions: )
No matching distribution found for dbus

**Running:**
sudo apt install python-dbus

**I get:**

Reading package lists... Done
Building dependency tree
Reading state information... Done
python-dbus is already the newest version (1.2.4-1).
0 upgraded, 0 newly installed, 0 to remove and 165 not upgraded.

So I assumed I was have it installed properly.

**I invoke the script with:**
sudo python /home/pi/card_game/pyplayer.py

as for sys.path & sys.executable

**sys.path:**
/home/pi
/usr/lib/python2.7
/usr/lib/python2.7/plat-arm-linux-gnueabihf
/usr/lib/python2.7/lib-tk
/usr/lib/python2.7/lib-old
/usr/lib/python2.7/lib-dynload
/usr/local/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages/gtk-2.0

**sys.executable:**
/usr/bin/python

I hope this is all the right information. Thank you for a quick response!

**The full code of the script:**
#!/usr/bin/env python

from omxplayer.player import OMXplayer
from time import sleep

player = OMXplayer('/home/pi/card_game/video/reward1.mp4')

player.play()
sleep(5)
player.pause()

player.quit()

willprice, 2018-01-17:
Thanks for the information, how did you install `python-omxplayer-wrapper`?

landtree, 2018-01-17:
I used the pip line on the docs site:
pip install omxplayer-wrapper

willprice, 2018-01-17:
You're missing `/usr/lib/python2.7/site-packages` in `sys.path` which is where python usually installs things. It looks like you have an issue with pip or your python install. Can you install anything else with `pip` and import it? e.g. `numpy`?

landtree, 2018-01-17:
I installed pyserial from pip and I am using that in a different script that works fine.

willprice, 2018-01-17:
Interesting, well we need to find out where `omxplayer` installed, if at all. Could you attach the logs generated by running:

```
$ tree .local/lib/python* > user-lib.log
$ tree /usr/lib/python* > system-lib.log
```

nellapatella, 2018-01-18:
I have the exact same problem as landtree.
I installed everything as instructed but get the same ImportError.  

Any ideas yet what to do?

willprice, 2018-01-18:
@nellapatella if you can provide the output of the tree commands then I can help diagnose the issue

nellapatella, 2018-01-18:
For some reason when I do the tree commands I get nothing...  
Sorry, super new to all of this

willprice, 2018-01-18:
@nellapatella can you post the exact output of the command? If you haven't got tree installed run `sudo apt-get install tree`

nellapatella, 2018-01-18:
I get this... nothing else...
And tree is installed.

pi@raspberrypi:~ $ tree .local/lib/python2.7 > user-lib.log
pi@raspberrypi:~ $ tree /usr/lib/python2.7 > system-lib.log
pi@raspberrypi:~ $

willprice, 2018-01-18:
Yes, that is correct, I now need the `user-lib.log` and `system-lib.log` files to be attached to the ticket so I can look at them, `>` redirects output to the terminal into a file.

nellapatella, 2018-01-18:
ah!  sorry had no idea that was happening.  Here they are.  [
[user-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/1642405/user-lib.log)
[system-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/1642406/system-lib.log)

nellapatella, 2018-01-18:
Maybe its because I am using Python 3?  I read some other threads here that had this issue...?

willprice, 2018-01-18:
That would almost certainly be the issue as you have used pip2 to install the package looking at the logs you've provided. Try `pip3 install omxplayer-wrapper` if you want to install for python 3.

nellapatella, 2018-01-18:
Awesome!  Got it working!  Thanks for such a quick response!

landtree, 2018-01-18:
Sorry for the late reply, I had a few meetings and then stepped out of the office.

I ran the tree commands. 

[system-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/1643823/system-lib.log)
[usr-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/1643824/usr-lib.log)

willprice, 2018-01-19:
I cannot see `omxplayer` installed anywhere in your python tree... can you try installing it again with `sudo pip2 install omxplayer-wrapper`

jamesm330, 2018-10-08:
Hello,
I too am having the same problems getting the example to work at all.  I get The same "No module named 'omxplayer'"

I am running Python3.5.3 on a Rpi Model2.  Any help on this is greatly appreciated as this is my only hope at getting a Python program to control OMXplayer.

Here are the log files:
[system-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/2457250/system-lib.log)
[user-lib.log](https://github.com/willprice/python-omxplayer-wrapper/files/2457251/user-lib.log)

willprice, 2018-10-08:
How about the output `python3 -c 'import sys; print(sys.path)'`. You have the library installed locally for both 2.7 and 3.5

jamesm330, 2018-10-08:
Thats possible as I followed the instructions and when it did not work found this thread and saw the mention of Python 3 and the install method for that.  I then tried that so yes I may have it installed twice.  Should I remove the Python 2.7 install, and if yes how do I do that?  Sorry for the noob-ness, its because at this I am a noob.

willprice, 2018-10-09:
If you provide the output of
`python3 -c 'import sys; print(sys.path)'` I might be able to help you pinpoint why you can't import the library.

jamesm330, 2018-10-09:
Here is what is reported back:

[python3_info.txt](https://github.com/willprice/python-omxplayer-wrapper/files/2460368/python3_info.txt)

willprice, 2018-10-09:
Are you launching python with `sudo` or as a user different to `pi`?
You have the library installed locally for your `pi` user, but not globally, so if you execute python with `sudo` your `PYTHONPATH` won't have the directory containing the `omxplayer` lib on it. I suggest you try installing globally if you're executing with a different user `sudo pip install omxplayer-wrapper`. If you're using a virtualenv you'll have to reinstall the package in there.

jamesm330, 2018-10-10:
I am launching python within IDLE at the moment as I am still putting the application together.  The programming is being done on the Rpi itself.  How do I install globally?  And do you mean install OMXPLAYER, or the Wrapper?

jamesm330, 2018-10-10:
Yes, I do have it installed.  I tried installing it with sudo pip install omxplayer-wrapper which failed, and I then tried sudo pip3 install omxplayer-wrapper, which did install

willprice, 2018-10-10:
How did the install fail?
What are the outputs of these commands?

`python -c 'import omxplayer; print(omxplayer.__version__)'`
`python3 -c 'import omxplayer; print(omxplayer.__version__)'`

jamesm330, 2018-10-10:
I ran this:
sudo pip install omxplayer-wrapper

again and this time it did not give back an error....whatever
I ran this:
python -c 'import omxplayer; print(omxplayer.__version__)'
And this came back:
`pi@raspberrypi:~ $ python -c 'import omxplayer; print(omxplayer.__version__)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python2.7/site-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/home/pi/.local/lib/python2.7/site-packages/omxplayer/player.py", line 18, in <module>
    from dbus import DBusException, Int64, String, ObjectPath
  File "dbus.py", line 5, in <module>
    import pathlib
ImportError: No module named pathlib
`

I ran this:
python3 -c 'import omxplayer; print(omxplayer.__version__)'

and this came back:
`pi@raspberrypi:~ $ python3 -c 'import omxplayer; print(omxplayer.__version__)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/home/pi/.local/lib/python3.5/site-packages/omxplayer/player.py", line 18, in <module>
    from dbus import DBusException, Int64, String, ObjectPath
  File "/home/pi/dbus.py", line 7, in <module>
    from omxplayer.player import OMXPlayer
ImportError: cannot import name 'OMXPlayer'
pi@raspberrypi:~ $ 
`

THoroughly confused at this point

willprice, 2018-10-10:
I believe your file is called `dbus.py`? Is that right? Don't call it that, it conflicts with one of the built-in modules causing a circular import, rename it to something else. If this isn't your file, just delete it so it doesn't exist at `/home/pi/dbus.py`.

closed 2018-10-17 by willprice

willprice, 2018-10-17:
> Have been ill and not at work on this sorry.

Feel free to reopen if it's an issue.

supremeinternational, 2020-05-01:
hi i need your help .   from omxplayer.player import OMXPlayer
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
  File "/home/pi/.local/lib/python3.5/site-packages/omxplayer/player.py", line 18, in <module>
    from dbus import DBusException, Int64, String, ObjectPath
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
  File "/home/pi/dbus.py", line 2, in <module>
    from omxplayer.player import OMXPlayer
ImportError: cannot import name 'OMXPlayer'