willprice/python-omxplayer-wrapper #118: omxplayer wrapper as child tkinter frame

stevanusadi, 2018-01-25:
hi, Sir..
i make an python application and i need to add my video in this python.
my python application using 2 main frame,    frame in left and right monitor. i wonder, can i add omxplayer to play video file in one of this frame, so omxplayer to be a child of a frame? 
thank you.

best regards,
Stevanus Adi

willprice, 2018-01-25:
Sorry, no idea. This wrapper just launches an omxplayer process, so if you can figure out how to embed the window launched by omxplayer on the command line into a tkinter window, then you can inject the same args into the constructor of the OMXPlayer object

stevanusadi, 2018-01-25:
ok Sir.. thanks you.

sincerely yours..

innlight, 2020-03-12:
stevanusadi  i have the same problem. Have you resolved it?