willprice/python-omxplayer-wrapper #119: Random hangs using USB audio (ALSA)

clickworkorange, 2018-01-26:
My python-omxplayer-wrapper implementation randomly hangs, and audio often lags severely behind video. This tends to happen more frequently when the Pi has been left alone for a little while (a couple of minutes), with OMXPlayer in the paused state. I'm using a USB soundcard via ALSA. The issue *seems* to go away if I unplug the soundcard and instantiate OMXPlayer without the `-o alsa` argument. The Python code looks like this: 

```python
import os
import RPi.GPIO as GPIO
from omxplayer import OMXPlayer

os.system("echo 1 | sudo tee /sys/class/backlight/rpi_backlight/bl_power")

player = OMXPlayer("movie.mp4", args=['--no-osd', '--loop', '--aspect-mode', 'stretch', '-o', 'alsa', '--vol', '-600'], pause=True)

def play_pause(pin):
    if(GPIO.input(pin) == 0):
        os.system("echo 0 | sudo tee /sys/class/backlight/rpi_backlight/bl_power")
        player.play()
    else:
        player.set_position(0)
        player.pause()
        os.system("echo 1 | sudo tee /sys/class/backlight/rpi_backlight/bl_power")

GPIO.setmode(GPIO.BOARD)
GPIO.setup(37, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.add_event_detect(37, GPIO.BOTH, bouncetime=500)
GPIO.add_event_callback(37, play_pause)

try:
    while True:
        pass

except KeyboardInterrupt:
    player.quit()
    GPIO.cleanup()
    os.system("echo 0 | sudo tee /sys/class/backlight/rpi_backlight/bl_power")
```

## Environment details
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.3 (stretch)
Release:	9.3
Codename:	stretch

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.5` |
| `python-dbus`  |  `1.2.4-1` |
| `python-evento`  |  `1.0.1` |
| `python-decorator`  |  `4.2.1` |
| `python-RPi.GPIO`  |  `0.6.3` |
| `python` | `2.7.13` |
| `omxplayer` | `5a25a57` |

jaco01, 2018-07-06:
I have hangs as well if I use '-o' or '--adev' arguments Regardless of value of arguments (hdmi, both, local, alsa). Additionally, omxplayer won't exit player.sync if using audio output arguments

willprice, 2018-11-03:
>  audio somtimes lags severely behind video
That sounds like an issue with `omxplayer`, not this library. Can you replicate the issue with `omxplayer` on the command line or does it only appear when using this library?

rbckman, 2019-04-18:
ok, I can confirm that this happens, raspberry pi will hang (zombify the process), nothing else will work but to restart the pi to shut it down, but as Will is pointing out this is probably a fault in omxplayer in it self.. I'm bypassing this problem by playing the audio file separately with aplay, works quite well.