willprice/python-omxplayer-wrapper #123: Not able to open two players at once and hide one video

houtanb, 2018-06-30:
# Issue Report
## Description
I try to open two players. I hide both then play one. After a few seconds, both videos are flashing on the screen when I should never see the video in `player1`.

## Problem reproduction
```
import time
from omxplayer.player import OMXPlayer

player0 = OMXPlayer('a.mp4', args=['--no-osd'])
player0.pause()
player0.hide_video()

player1 = OMXPlayer('b.mp4', args=['--no-osd'])
player1.pause()
player1.hide_video()

time.sleep(2)

player0.show_video()
player0.play()

time.sleep(2)

player0.pause()
player0.hide_video()

player0.quit()
player1.quit()
```

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.4-1` |
| `python` (`python --version`) | `2.7.13` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

sevildevil1990, 2018-06-30:
Hi,
try using the `--layer` argument on both of the players, with `args=['--layer','2']` for player0 and `args=['--layer','1']` for player1.

houtanb, 2018-06-30:
@sevildevil1990 Thanks so much! So for me the limit seems to be 3 videos as on playing the fourth the screen goes black. I guess this is hitting the limits of the graphics card? Do you have any suggestions for increasing the number of videos that can be played? I've bumped my graphics card memory up to 500MB, which is far more than the several MB that the videos consume.....
```
import time
from omxplayer.player import OMXPlayer

names = 'a.mp4', 'b.mp4', 'c.mp4'

player0 = OMXPlayer('a.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(0))
player0.pause()
player0.hide_video()

player1 = OMXPlayer('b.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(1))
player1.pause()
player1.hide_video()

player2 = OMXPlayer('c.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(2))
player2.pause()
player2.hide_video()

player3 = OMXPlayer('d.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(3))
player3.pause()
player3.hide_video()

player4 = OMXPlayer('e.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(4))
player4.pause()
player4.hide_video()

player5 = OMXPlayer('f.mp4', args=['--no-osd'], dbus_name='org.mpris.MediaPlayer2.omxplayer' + str(5))
player5.pause()
player5.hide_video()

time.sleep(2)

player0.show_video()
player0.play()

time.sleep(2)

player0.pause()
player0.hide_video()

player1.show_video()
player1.play()

time.sleep(2)

player1.pause()
player1.hide_video()

player2.show_video()
player2.play()

time.sleep(2)

player2.pause()
player2.hide_video()

player3.show_video()
player3.play()

time.sleep(2)

player3.pause()
player3.hide_video()

player4.show_video()
player4.play()

time.sleep(2)

player0.quit()
player1.quit()
player2.quit()
player3.quit()
player4.quit()
player5.quit()

```

sevildevil1990, 2018-07-01:
@houtanb I think we're running more into issues with the Raspi's overall performance.

On the button press: Will there be buttons to access all of the videos at random, or will it be more like a 'next' button? For the last case, you could only use two players: One that is playing the current video and one that already loads the next video into the memory. While pressing the button you would start the already loaded video, stop the previous player (and by that killing it's process) and then prepare the player for the next video in the queue.

houtanb, 2018-07-01:
@sevildevil1990 Thanks for taking the time to respond. But, in effect, I am looking for something more like the former. 3 buttons and 6 videos: one video for button press and one for button release. And there's no knowing what action will come ahead of time

houtanb, 2018-07-01:
I will close this issue as it is not a bug with the `python-omxplayer-wrapper`

closed 2018-07-01 by houtanb

houtanb, 2018-07-11:
First, please enclose code in "```". See: [Github Markdown](https://guides.github.com/features/mastering-markdown/)

Second, I don't quite understand what you're saying. But the Raspberry Pi has trouble supporting more than one `omxplayer` at a time.

The solution for me was to link all the videos together then jump to different positions depending on what I wanted to play.

MakeProjects, 2018-07-11:
My apologaize for my mistake @houtanb, I´ve fixed the last post. It is the first time that I write on github. I hope my problem help to someone to. 

I need to switch betewen different videos, I tried different options with omxplayer-wrapper but I always got a black screen during the videos are switching. I need to switch between video1 and video2 without black screen and within delay.  

I think that you solution is the best. Put the videos together and then jump to de the position. 

Could you show me that you code this?

Thanks a lot.