willprice/python-omxplayer-wrapper #124: .pause() stops audio playback on all running inscances

sevildevil1990, 2018-06-30:
## Description
Hi,
I currently try to make the playback of multiple videos more seamless by starting a second player instance just ahead of the end of the first video, pausing it right away and eventually start the playback as soon as the first video finishes.
This works well for the video part! The audio however stops on both players, as soon as the second player is paused – the video keeps playing though.

## Problem reproduction
```
#!/usr/bin/env python3
import time
from omxplayer.player import OMXPlayer

print("start player1")
player1 = OMXPlayer("/home/pi/Videos/video1.mp4", args=['--layer','2'], dbus_name="org.mpris.MediaPlayer2.omxplayer1")
time.sleep(2)

print("load player2")
player2 = OMXPlayer("/home/pi/Videos/video2.mp4", args=['--layer','1'], dbus_name="org.mpris.MediaPlayer2.omxplayer2")
player2.pause()
# Audio cuts off for both right here

time.sleep(2)
player1.stop()
player2.stop()
```

I tried multiple scenarios now:
- If I start player2 without pausing it, the audio blends just fine. Just when I try to pause *any* of the players, the audio cuts off for *all* of them.
- Muting and Unmuting player1 after player2 is paused still left player1 seemingly muted.
- Pausing and Unpausing player1 after player2 is paused did work, but created a little audio stutter as you would expect. 
- This issue seems to affect both, python2 and python3 environments.

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.4 (stretch)
Release:        9.4
Codename:       stretch
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.5` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `2.7.13`/`3.5.3` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

willprice, 2018-11-03:
I think this is an issue with `omxplayer` itself, I can't run two instances of omxplayer and pause one without pausing the other either.
Try running in two terminal windows and running the two commands:

```
# terminal 1
$ omxplayer tests/media/test_media_1.mp4 --layer 1 --dbus_name org.mpris.MediaPlayer2.omxplayer1 --win '0 0  200 200'

# terminal 2
$ omxplayer tests/media/test_media_1.mp4 --layer 2 --dbus_name org.mpris.MediaPlayer2.omxplayer2 --win '200 0  400 200'
```

When I pause one it pauses the other player.

Closing as I can't fix this, it's an issue with omxplayer itself, sorry about that.

closed 2018-11-03 by willprice

Eskimon, 2019-02-26:
Hello and thanks for the great work.

I just want to give some input there. I think the problem is rather on the dbus side. If you play an omxplayer instance in two different terminals, and play/pause with the 'p' key it's working just fine, you can control any of the omxplayer instance independently.

juanmartin, 2019-10-01:
You should rename the DBUS process in order to be able to control them independently
Like so:
```
player = OMXPlayer(video, args=yourArgs, dbus_name=org.mpris.MediaPlayer2.omxplayer1)
player = OMXPlayer(video, args=yourArgs, dbus_name=org.mpris.MediaPlayer2.omxplayer2)
```
Hope it helps.