willprice/python-omxplayer-wrapper #125: Issue Report: DBus cannot connect to the OMXPlayer process

AidanTek, 2018-07-16:
# Issue Report
## Description
I can't run the basic examples included in the documentation, I get the "SystemError: DBus cannot connect to the OMXPlayer process". I can see this is a historical issue but I can't find any way to resolve it. I have followed the installation as described in docs - I'm using Python 3 so I used pip3 install etc. Also tried in Python 2 with no luck.

## Problem reproduction
```
from omxplayer.player import OMXPlayer
from pathlib import Path
from time import sleep

FILE = '../testmedia/test0.mp4'

player = OMXPlayer(FILE) # no difference with argument dbus_name='org.mpris.MediaPlayer2.omxplayer0'

sleep(5)

player.quit()
```

## Environment details

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.5` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.4-1` |
| `python` (`python --version`) | `3.5.3` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

-------

himijendrix24, 2018-08-31:
For me missing permissions on the /tmp folder caused this issue.
Something on raspbian must have changed, because in the past it never was a problem.

A quick fix was adding this to /etc/rc.local
sudo chmod 1777 /tmp

clacktronics, 2018-10-18:
Omxplayer has changed in stretch? it is now a fork based on this repository.

`omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Wed, 05 Jul 2017 16:40:39 +0100
        Version   : 5a25a57 [debian]
        Repository: git@github.com:XECDesign/omxplayer.git
`
https://github.com/XECDesign/omxplayer/commit/5a25a579a81a585c5166cd4896e8e8a56180d6e8

I am trying to find out what has changed

jehutting, 2018-10-19:
I was surprised to find out about [this XECDesign release](https://www.raspberrypi.org/forums/viewtopic.php?t=224110).

Apparently the official omxplayer releases comes no longer from the popcornmix repository but from the XECDesign repository branch debian.

willprice, 2018-11-03:
The response by @johannesvogel is correct. You'll need to pass in a `Path` object, I'll fix this in https://github.com/willprice/python-omxplayer-wrapper/issues/149 so you can also pass in a string.

closed 2018-11-03 by willprice

Kaeosm, 2019-08-20:
> I also got this error message when I ran my own script. I figured out the problem was, that I wasn't using pathlib.
> 
> After changing this
> 
> ```python
> from omxplayer.player import OMXPlayer
> 
> path = 'path/to/file'
> player = OMXPlayer(path)
> ```
> 
> to this
> 
> ```python
> from omxplayer.player import OMXPlayer
> from pathlib import Path
> 
> path = Path('path/to/file')
> player = OMXPlayer(path)
> ```
> 
> the error doesn't occur anymore for me. I see that this is the same for @AidanTek.

Thanks for contributing - I was able to get the DBus error to cease - However, I have to remove all arguments to achieve a video playing successfully.

Is there an exception to adding arguments using this method? I am attemping to utilize '--loop' and some window/aspect arguments as well through hdmi.

I'm utilizing @JonnyAlpha 's method, and once I get the looping video, I will be utilizing GPIO Pull Down/Up to initiate a 2nd video to play, and upon the 2nd videos completion - Start the Looping video again.

I'm using Any input is greatly appreciated.
``` 
import os
import sys
import keyboard
from time import sleep
from omxplayer import OMXPlayer
from signal import pause
from pathlib import Path

##from keyb import KBHit
##kb = KBHit()

header = 2
loop = '/home/pi/Videos/test.h264'
donation = '/home/pi/Videos/Trailer.mp4'

videos = './'
adev='hdmi'
vid1 = OMXPlayer(videos+loop,args=['--no-osd', '--no-keys', '--win', '100 100 640 480', '--loop', '-o', adev], dbus_name='org.mpris.MediaPlayer2.omxplayer1', pause=True)

def loop():
        initiate()
        print("Playing Loop")
        vid1.play()

def donation_success():
    while True:
        header_state = GPIO.input(header)
        if header_state == GPIO.HIGH:
            os.system('killall omxplayer.bin')
            print("Accepted!")
            sleep(2)
            print("Stopping Loop")
            vid1.stop()
            print("Playing Video, Thank You!")
            vid2 = OMXPlayer(videos+donation,args=['--no-osd', '--no-keys','-o', adev], dbus_name='org.mpris.MediaPlayer2.omxplayer2', pause=True)
            vid2.play_sync()

if __name__ == '__loop__':
   loop()
```