willprice/python-omxplayer-wrapper #134: Running as sudo causes crash

3djake, 2018-09-07:
# Issue Report
I cannot run program as root/sudo, it causes program to crash.

## Problem reproduction
Use any example script as a regular user and it works as expected.
Then run it with sudo and it will crash
`dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.`

Environment details
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
python-dbus Version 1.2.0-3
Python 3.5.2
omxplayer Version  dfea8c9 
```

3djake, 2018-09-08:
I do not know what happened but the issue seemed to resolve itself. I have to deploy my program on a few more pi's next week so I will try to figure out the cause and solution.

willprice, 2018-09-10:
Hmmm interesting. If it happens again check `/tmp` and see what `omxplayer*` files are present, it could be that one of the files that contains the dbus address didn't get cleaned up from a previous crash.