willprice/python-omxplayer-wrapper #148: NameError: name 'player' is not defined

AnZwecky, 2018-10-24:
# Issue Report
## Description
I wrote a basic program for controlling omxplayer via GPIO buttons through the wrapper. I'm using Popen to play a video file but when I test the play/pause button during playback using player.pause(), I get the error: "NameError: name 'player' is not defined". (I would use the wrapper for all control functions but it doesn't work.) I've checked countless posts on many different forums and tried many things that would likely have worked but too advanced for me, including attempting to integrate dbus manually but that's way over my head at this time. My skill level with the Pi and python is at the general hobbyist level so explicit code and terminal commands with simple instructions and explanations are ideal. I wanted a simple library to work with but it's become a headache....

## Problem reproduction
from subprocess import Popen
from omxplayer.player import OMXPlayer

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)

playPause = 14
play_led = 2

GPIO.setup(playPause, GPIO.IN)
GPIO.setup(play_led, GPIO.OUT)

vFile = '/media/pi/MEDIACARD/*.mp4
cmd = "omxplayer %s" %(vFile)

while True:
    if GPIO.input(playPause) == False:
        time.sleep(0.2)
        Popen([cmd], shell = True)
        GPIO.output(play_led, True)
        if GPIO.output(playPause) == False:
            player.pause()

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.4 (stretch)
Release:        9.4
Codename:       stretch
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.5` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.4-1` |
| `python` (`python --version`) | `2.7.13` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

jehutting, 2018-10-25:
Hi,

You are mixing some things up :-)

You don't need ```Popen```. All that kind of stuff is wrapped in the OMXPlayer wrapper.

The reason for the ```NameError: name 'player' is not defined``` error is that you haven't created _an instance_ called ```player``` of the (wrapper) class OMXPlayer:
``` 
player = OMXPlayer(vFile)
```
Once having this instance you can control the real OMXPlayer with the wrapper instance functions like 
```
player.pause()
player.play() 
player.quit()
```

In your case you also need the ```is_playing()``` function. With this function you can determine if the player is playing or not. So when you press your button ```playPause``` check the status and decide from its value to either call ```pause()``` or ```play()```.

AnZwecky, 2018-10-25:
The reason I didn't use `player = OMXPlayer(vFile)` to start was because I tried but upon pressing my play button I got `SystemError: DBus cannot connect to the OMXPlayer process`. (I tried again for kicks but the issue remains.) I tried finding a fix for it but all the posts I read went way over my head.

willprice, 2018-11-03:
Thanks for the comments @jehutting

closed 2018-11-03 by willprice