willprice/python-omxplayer-wrapper #157: player.py requires mock, but it is not included with lib_deps

superlou, 2018-11-10:
# Issue Report
## Description
The [example](https://python-omxplayer-wrapper.readthedocs.io/en/latest/) in the docs fails if mock isn't installed on the user's system. This is fixed by installing mock. Mock is included with the test dependencies, but not `lib_deps`. I'd create a PR to simply add it to the installation dependencies, but I'm not sure if the intent is to require mock for all users.

## Problem reproduction
```
sudo pip3 install omxplayer-wrapper
python3
>>> import omxplayer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.5/dist-packages/omxplayer/player.py", line 10, in <module>
    from mock import Mock
ImportError: No module named 'mock'
```

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID:	LinuxMint
Description:	Linux Mint 18.3 Sylvia
Release:	18.3
Codename:	sylvia
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `1.11.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-3` |
| `python` (`python --version`) | `3.5.2` |
| `omxplayer` (`omxplayer --version`) | `(currently starting up on Linux Mint and moving to RPi once the application's testable features are ready)` |
-------

willprice, 2018-11-10:
OK, v0.3.2 is out, should fix this, let me know if not.

closed 2018-11-10 by willprice

superlou, 2018-11-10:
That fixed it, thanks!