willprice/python-omxplayer-wrapper #161: One player for multiple videos

Tastyep, 2018-11-26:
# Feature
## Description
Add a default constructor to the OMXPlayer object and allow the user to pass options to the load() method.

## Example use case
```python
player = OMXPlayer() # default constructed player

for video in videos:
    player.load(video.path, ['--subtitles', video.subtitle])
```

The example is extremely simplified, the videos should not be played in a loop like this and I don't expect the load() method to be blocking.
The idea is simply to not have to recreate a player object every time a video needs to be played.

samshum2, 2020-08-28:
sorry , I can't run the program  in the first line:
TypeError: __init__() missing 1 required positional argument: 'source'

samshum2, 2020-08-28:
player1 = OMXPlayer(Video1,dbus_name='org.mpris.MediaPlayer2.anyName')
player2 = OMXPlayer(Video2,dbus_name='org.mpris.MediaPlayer2.anyName2')
player2.pause()
sleep(player2.duration())
player2.play()

ShaoMingya, 2020-11-18:
> player1 = OMXPlayer（Video1，dbus_name ='org.mpris.MediaPlayer2.anyName'）
> player2 = OMXPlayer（Video2，dbus_name ='org.mpris.MediaPlayer2.anyName2'）
> player2.pause（）
> sleep（player2.duration（））
> player2 。玩（）

playerN = ... MAX_N?