willprice/python-omxplayer-wrapper #167: Ability to set loop after playback started

superlou, 2018-12-15:
# Feature Request
## Description
Is it possible to set OMXPlayer's "--loop" argument after playback has already started?

## Example use case
I am working on a playout system for live video production, where a Raspberry Pi is playing video to a hardware video switcher. While the video is playing, the operator may decide that the video needs to be looped in order to run long enough. My understanding is that the only way to get seamless playback from OMXPlayer is to set the "--loop" argument. Trying to catch the stop-playback event and seek to the beginning causes gaps.

However, I'm not sure that whatever flag "--loop" sets is exposed over the dbus interface, so this might simply not be possible. I've been trying to find a hack where I always launch the player with "--loop" set, but I haven't been able to reliably determine when the video is at the end of playback and stop it before showing some of the beginning again.

willprice, 2018-12-15:
Hi There,

Sadly I don't believe this is implemented in omxplayer, so can't be exposed in this library. All the implemented methods can be found in https://github.com/popcornmix/omxplayer/blob/master/OMXControl.cpp

closed 2018-12-15 by willprice

superlou, 2018-12-16:
Thanks, I was afraid of that. I've opened an issue on [OMXPlayer](https://github.com/popcornmix/omxplayer/issues/668) to figure out what it would take to add to the player.

willprice, 2018-12-16:
Cool, let me know if it ever gets implemented as it would be trivial to add once exposed.
Cheers.