willprice/python-omxplayer-wrapper #168: Can't install omxplayer-wrapper on Raspbian

chq1976, 2018-12-16:
Hi,

i have a brandnew installed "Raspbian with desktop".

When i type in terminal..

sudo apt-get update && sudo apt-get install -y libdbus-1{,-dev}

..i geht this errormessage (translated from german):

Package libdbus-1 can not be found.

What could be the problem?

Greetz Chris

willprice, 2018-12-16:
Hi Chris,

Looks like `libdbus-1` no longer exists in the latest release, try `sudo apt install -y libdbus-1-3 libdbus-1-dev`

pwobus, 2021-02-24:
i also have not been able to install omxplayer-wrapper on either a pi zero or pi4 running a fresh install of raspbian.
I used pip and got the libdus missing error message,  I then tried installing it again after sudo apt install -y libdbus-1-3 libdbus-1-dev.
This error message appeared:  No package 'glib-2.0' found and
command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-3H5M9M/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-WBOnBU/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-3H5M9M/dbus-python/

Hoping to find an alternative to omxplayer-wrapper.

willprice, 2021-02-24:
It looks like you're missing glib2, try install that through apt

pwobus, 2021-02-25:
I did sudo apt-get install libglib2.0-dev then sudo pip install omxplayer-wrapper
Everything appeared to install correctly.Successfully built dbus-python
Installing collected packages: dbus-python, decorator, evento, scandir, pathlib2, omxplayer-wrapper
Successfully installed dbus-python-1.2.16 decorator-4.4.2 evento-1.0.2 omxplayer-wrapper-0.3.3 pathlib2-2.3.5 scandir-1.10.0

But still something wrong when I use Omxplayer-wrapper
ModuleNotFoundError: No module named 'omxplayer.player'; 'omxplayer' is not a package