willprice/python-omxplayer-wrapper #180: No module named 'decorator'

Britican, 2019-03-14:
# Issue Report
## Description
After installing omxplayer.wrapper I get the following import error: 

from omxplayer.player import OMXPlayer
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/player.py", line 17, in <module>
    from decorator import decorator
ImportError: No module named 'decorator'

## Problem reproduction
I used 'sudo pip3 install omxplayer-wrapper' to install
In the Python Shell:
from omxplayer.player import OMXPlayer

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.3.2` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `3.4.2` |
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

willprice, 2019-03-14:
That's weird, it looks like your installation of `decorator` failed, this is a dependency of this module and is listed in the `setup.py`---so this should have been installed when you ran `pip install omxplayer-wraper`.

What happens when you run
```
$ sudo pip3 install decorator
```

Britican, 2019-03-14:
Thanks for the speedy reply..

That worked, no decorator error now, however, now i get the following new error:

from omxplayer.player import OMXPlayer
Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/player.py", line 18, in <module>
    from dbus import DBusException, Int64, String, ObjectPath
ImportError: No module named 'dbus'

This is the information reported after requesting dbus version information

dpkg -s python-dbus
Package: python-dbus
Status: install ok installed
Priority: optional
Section: python
Installed-Size: 415
Maintainer: Utopia Maintenance Team <pkg-utopia-maintainers@lists.alioth.debian.org>
Architecture: armhf
Source: dbus-python (1.2.0-2)
Version: 1.2.0-2+b1
Replaces: python2.4-dbus
Provides: python2.7-dbus
Depends: python (>= 2.7), python (<< 2.8), libc6 (>= 2.4), libdbus-1-3 (>= 1.5.12), libdbus-glib-1-2 (>= 0.78), libglib2.0-0 (>= 2.12.0), python-dbus-dev
Recommends: python-gi | python-gobject-2 | python-qt4-dbus
Suggests: python-dbus-doc, python-dbus-dbg
Breaks: gajim (<< 0.11.1), gnome-osd (<< 0.12.0), python-qt4-dbus (<< 4.8.3-3)
Conflicts: python2.4-dbus
Description: simple interprocess messaging system (Python interface)
 D-Bus is a message bus, used for sending messages between applications.
 Conceptually, it fits somewhere in between raw sockets and CORBA in
 terms of complexity.
 .
 This package provides a Python interface to D-Bus.

I don't see any mention of python3 which is what I'm using, is that something to do with it?

willprice, 2019-03-14:
The package is python 3 compatible, can you paste the output of `dpkg -s python3-dbus` and if that spits out an error, run `sudo apt install python3-dbus`

Britican, 2019-03-14:
Yes, it's not installed so I did the sudo apt install python3-sbus

It installed successfully, now I get this:

Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/__init__.py", line 1, in <module>
    from omxplayer.player import OMXPlayer
  File "/usr/local/lib/python3.4/dist-packages/omxplayer/player.py", line 25, in <module>
    from evento import Event
ImportError: No module named 'evento'

willprice, 2019-03-14:
It looks like none of the dependencies of the package were installed, can you try rerunning `sudo pip3 install omxplayer-wrapper`, if that doesn't fix your issues, how about `sudo python3 -m pip install omxplayer-wrapper`

Britican, 2019-03-14:
I tried both:

pi@raspberrypi:~ $ sudo pip3 install omxplayer-wrapper
Requirement already satisfied (use --upgrade to upgrade): omxplayer-wrapper in /usr/local/lib/python3.4/dist-packages
Cleaning up...

pi@raspberrypi:~ $ sudo python3 -m pip install omxplayer-wrapper
Requirement already satisfied (use --upgrade to upgrade): omxplayer-wrapper in /usr/local/lib/python3.4/dist-packages
Cleaning up...

willprice, 2019-03-14:
And if you uninstall?
```
$ sudo python3 -m pip uninstall omxplayer-wrapper
$ sudo python3 -m pip install omxplayer-wrapper
$ python3 -c 'import omxplayer
```
Does the last command execute successfully or also give import errors? If it's successful, you should be good to go.

Britican, 2019-03-14:
Just to confirm the syntax of the last command:
$ python3 -c 'import omxplayer
Correct, the single quote in front of import?

willprice, 2019-03-14:
Nope, you're quite right, that was incorrect, I've edited it adding a single quote at the end too, so the command should be
```
$ python3 -c 'import omxplayer'
```

Britican, 2019-03-14:
The last three commands fixed the problem, no more import errors :O)

Thank you very much for your help, I'm looking forward to developing with the omxplayer-wrapper!

Thanks again!

willprice, 2019-03-14:
No worries, glad it worked :)

closed 2019-03-17 by willprice