willprice/python-omxplayer-wrapper #183: OMXPlayer instantiation halts execution when in Flask route handler

superlou, 2019-07-11:
# Issue Report
## Description
I just updated from Stretch to Buster, and I'm having a problem (but "not" exception) when calling `OMXPlayer(...)`. I say "not" because in a very specific usage, while no exception is generated, code after the call does not continue. This only seems to happen when trying to create the player in a Flask route handler, but this was working on Stretch and I haven't found any documentation indicating this behavior would change.

## Problem reproduction
I have created a helper class to manage my communication with OMXPlayer in a Flask webserver.

```
class Omx:
   ...
        
    def play(self, filename):
        if self.player:
            self.expects_loading_exit = True
            self.player.load(filename)
        else:
            print('here1')
            self.player = OMXPlayer(filename, args=['-b', '--no-osd', '-o', 'both']) #, '--loop'])
            print('here2')
            self.player.stopEvent += self.on_player_stop
            self.player.exitEvent += self.on_player_exit
```

The full project is [on github](https://github.com/superlou/pideck).

If I instantiate this class and run in a Python 3 shell, everything works: the video plays and "here1" and "here2" are printed. If the web server calls `play` during configuration, it also works with the same behavior:

```
def main():
    app.config['MEDIA_FOLDER'] = 'media'
    create_folder(app.config['MEDIA_FOLDER'])
    app.config['omx'] = Omx(app.config['MEDIA_FOLDER'])
    app.config['omx'].play('media/example.mp4')
    app.config['DEBUG'] = True

    socketio.run(app, host='0.0.0.0', port=8910, debug=True)
```

However, if I call `play` from inside the route, "here1" is printed and the video plays, but "here2" is not printed.

```
@app.route('/api/player', methods=['GET', 'POST'])
def player_status():
    if request.method == 'GET':
        omx = app.config['omx']
        omx.play('media/example.mp4')
```

Additional errors occur later (though not in the `play` function) as nothing after `print("here2")` is executed.

The weirdest thing is that no exception is printed. I can `raise Exception('test')` before initializing the `OMXPlayer`, and I see the exception, but nothing from or after initialization.

Is there anything in python-omxplayer-wrapper that might explain this behavior? I'm starting to think this might be more Flask related than I initially thought, but I'm trying to leave no stone unturned. Is there anything on Buster that could cause this? This approach worked under Stretch, though admittedly it's a fresh install, so it's possible that this is a symptom of something else.

## Environment details
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.3.2` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.8-3` |
| `python3-dbus` (`dpkg -s python3-dbus`) |  `1.2.8-3` |
| `python` (`python --version`) | `3.7.3` |
| `omxplayer` (`omxplayer --version`) | `f06235c` |

superlou, 2019-07-11:
Using PDB, and stepping over the `OMXPlayer(...)` call, I believe the error is a failure to connect to the OMXPlayer process.

```
> /usr/local/lib/python3.7/dist-packages/omxplayer/player.py(154)__init__()->None
-> self.load(source, pause=pause)
(Pdb) l
149  	        #: Event called on setting position ``callback(player, absolute_position)``
150  	        self.positionEvent = Event()
151  	
152  	        self._process = None
153  	        self._connection = None
154  ->	        self.load(source, pause=pause)
155  	
156  	    def _load_source(self, source):
157  	        if self._process:
158  	            self.quit()
159  	
(Pdb) s
SystemError: DBus cannot connect to the OMXPlayer process
> /home/pi/workspace/pideck2/omx.py(18)play()
-> self.player = OMXPlayer(filename, args=['-b', '--no-osd', '-o', 'both']) #, '--loop'])
```

superlou, 2019-07-15:
It definitely has something to do with executing in gevent. I switched to eventlet and it's working. I'm going to close this since I think it's a problem with my understanding of gevent, and not something in OMX player wrapper's DBUS access itself.

closed 2019-07-15 by superlou

jonra1993, 2019-11-11:
@superlou  How did you solve it ??

superlou, 2019-11-11:
@jonra1993, unfortunately, I was never able to figure out how to fix it in gevent. You can see the app that runs on eventlet here: https://github.com/superlou/pideck. The connection happens at https://github.com/superlou/pideck/blob/master/omx.py#L25.

jonra1993, 2019-11-11:
Thanks @superlou. I am not sure what happened with my python script but I clone omxplayer examples and they worked so I decided just edit examples.