willprice/python-omxplayer-wrapper #185: Run 2 videos consecutively

alexis85120, 2019-08-01:
Hi Will,

I would like to run 2 videos in loop at the same time and decide which one I want to appear on my screen. I want a very quick transition in order to be as real video. You'll understand with this code, I really simplified  that : 

if a=1
player1.pause() #Show only the video 2
player2=OMXPLAYER(Video2) #LOOP 
else
player2.pause #Show only the video 1 
player1=OMXPLAYER(Video1) #LOOP

sighmon, 2019-08-06:
@alexis85120 I've been using the `exitEvent` callback to start a second video, but unfortunately there's ~1 second between the two, which is causing my screen to flash. So I'm keen to see how you solve it. Here's my sample code:

```python
def omxplayer_exit_event_callback(self, player, exit_status):
    start_media_player()
end

omxplayer = OMXPlayer(video_file), args=' '.join(player_args), dbus_name=dbus_name)
omxplayer.exitEvent = omxplayer_exit_event_callback
```

alexis85120, 2019-08-06:
Hi @sighmon, I tried to use that too but it does not interest me because between the 2 videos I can't have more than 0.1 second, because it must be fluid. So, I did some researches and I found something. First, you have to assembly the 2 videos (the first from 0 to 30 and second to 30 to 50 for example). WIth that you can use player.seek(n) and player.set_position(n) to jump in the videos where ever you want. n is in seconds. player.seek will add n seconds to the videos ( For example if the videos is at 7 seconds and you put n=5 you'll have the videos to 13s. set.position() enable you to choose exactly a second from 0, no add. I hope I have been clear for you, you can look at this for more information : https://buildmedia.readthedocs.org/media/pdf/python-omxplayer-wrapper/latest/python-omxplayer-wrapper.pdf. 
Actually it works for me but it is not very precise, i will update if I find something perfect.

alexis85120, 2019-08-06:
If someone know how to use player.set_position(), Ill appreciate it. For me it does not work.

alexis85120, 2019-08-06:
@willprice