willprice/python-omxplayer-wrapper #189: Implement SetLayer method

JE-Random-Tech, 2019-08-31:
# Feature Request
## Description

For seamless video switching it would be ideal to be able to change the Layer of the Video while it is playing.

## Example use case
Setting the alpha channel to 255 or hide the video does not work as it always flickers therefor it is necessary to create a seamless video transition.

JE-Random-Tech, 2019-08-31:
I have already a working solution and would like to contribute it.

willprice, 2019-09-01:
Hi @JE-Random-Tech,

That sounds great. Just click the 'fork' button on the top right of this page to create your own copy of the repo. Once you've done that you can push your changes to the fork and open a pull request. The method should be in snake case for consistency with the rest of the API, i.e. `set_layer`.

Thanks :)

willprice, 2019-09-01:
Added in #191

closed 2019-09-01 by willprice