willprice/python-omxplayer-wrapper #194: python-omxplayer-wrapper Pi 4 dual video issue

zilogZ80, 2019-10-21:
I'm using the python-omxplayer-wrapper by willprice to control Omxplayer from Python, basically starting a video, pausing it until an input is received, playing it until the end, rewinding & pausing it, loop 

works great with one video but when I try to do it with two videos (both outputted via HDMI on a Pi 4) only the first video works as expected (i.e. pauses), the second one just plays normally. Any ideas? 

[code]
movie1 = ("/home/pi/zombie.mp4")
movie2 = ("/home/pi/sample video.mp4")

# start videos
# display=2 is HDMI 0
# display=7 is HDMI 1
omx_main = OMXPlayer (movie1, args=['-b', '--display=2', '-o', 'local'])
omx_sub = OMXPlayer (movie2, args=['-b', '--display=7'])
video_duration = omx_main.duration()
# pause videos
omx_sub.pause()
omx_main.pause()
[/code]

alexandrebarsacq, 2019-10-23:
What happens if you explicitly specify different dbus names, for example : 
`omx_main = OMXPlayer (movie1,dbus_name='org.mpris.MediaPlayer2.omxplayer1', args=['-b', '--display=2', '-o', 'local'])
`omx_sub` = OMXPlayer (movie2,dbus_name='org.mpris.MediaPlayer2.omxplayer2', args=['-b', '--display=7'])

zilogZ80, 2019-10-23:
Yes! Perfect, thank you. Apologies, it was probably really obvious, I am still learning though!!

closed 2019-10-23 by zilogZ80

willprice, 2019-10-23:
@alexandrebarsacq thanks for answering this. I've added a FAQ to the main page for the benefit of others like @zilogZ80 who have the same question
https://github.com/willprice/python-omxplayer-wrapper/blob/master/README.rst#how-do-i-create-multiple-players