willprice/python-omxplayer-wrapper #197: running under Alpine

kkhanh, 2019-11-15:
Hi,

I'm trying to use the library under Alpine. Do you know if there were any successful attempt ?

The steps I followed :
foo# pip3 setup.py install
foo# python3

>> from omxplayer.player import OMXPlayer
>> from pathlib import Path
>> VIDEO_PATH = Path("/tmp/test.mp4")
>> player = OMXPlayer(VIDEO_PATH)

From there I can see a new process on the list : 
2935 root      0:02 omxplayer /tmp/test.mp4

but nothing show up on the screen.

No problem when running omxplayer directly without python
Im using alpine-rpi-3.10.3-armhf

Any hints appreciated ty

willprice, 2019-11-15:
@kkhanh were you able to get this working? If so would you mind sharing your tips so others can benefit? Thanks

kkhanh, 2019-11-18:
@willprice I got it working by disabling openGL which is not enabled by default so there should be no problem for anyone. But the library is not working properly, /tmp is empty, looks like a problem with the dbus.

willprice, 2019-11-18:
Do you see any `*omxplayer*` files in `/tmp` when you run omxplayer from the CLI?

kkhanh, 2019-11-18:
@willprice No, /tmp is still empty. chmod 777 /tmp, same result