willprice/python-omxplayer-wrapper #198: Cannot install in PI4

t-walker-wei, 2019-12-28:
# Issue Report
## Description
Cannot install in PI4. show error:

    configure: error: Package requirements (dbus-1 >= 1.8) were not met:

    No package 'dbus-1' found

## Problem reproduction
    pip install omxplayer-wrapper

## Environment details
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.3.2` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.8-3` |
| `python` (`python --version`) | `2.7.16` |
| `omxplayer` (`exe`) | `6c90c75` |

-------

matthijskooijman, 2020-02-14:
Did you install `libdbus-1` and `libdubs-1-dev` with `apt` as [documented](https://python-omxplayer-wrapper.readthedocs.io/en/latest/#installation)?

TheOneWhoseUsernameWasTaken, 2020-08-29:
This is an older issue, but a just had a similar problem, so I wanted to mention it here.

On my Pi4 with Raspbian 10.4, I also had to install libglib2.0-dev (which in turn pulled a whole slew of other stuff). That wasn't documented.

```
    configure: error: Package requirements (glib-2.0 >= 2.40) were not met:
    No package 'glib-2.0' found
```

After installing libglib2.0-dev, the installation went through.

atoav, 2021-02-09:
I think this can be therefore closed now, I am testing on a fresh raspi now, if anything else pops up I will open a new issue/PR.