willprice/python-omxplayer-wrapper #199: player.pause() --> Type Error: 'bool' object is not callable

himijendrix24, 2020-02-11:
There seems to be a problem with the pause function (line 524) which also makes the load function (line 228) not working with the pause argument.
So somewehere the classses pause() function gets replaced by a boolean, but I can't find the problem in the code.

himijendrix24, 2020-02-11:
I think calling self.load(source, pause=pause) in line 154 should be self.load(source, pause)

matthijskooijman, 2020-02-14:
> I think calling self.load(source, pause=pause) in line 154 should be self.load(source, pause)

I don't think that would solve things, they would (with the current load method) be essentially be identical (`pause=pause` indicates to put the `pause` variable from the constructor and and assign it to the `pause` parameter of the load method, which also happens when you just pass it as the second argument).

Do you have a full backtrace of your exception?